luc-github/ESP32SSDP #1: SSDPClass::localIP() returns wrong IP if WiFi Mode is WIFI_AP_STA

MiTron66, 2019-04-25:
SSDPClass::localIP() retrung wrong IP if WiFi.getMode() == WIFI_AP_STA.

This leads to wrong ip in description.xml as this case is not covered. Device will not be detected in Windows 10.

A partially solution is to add this case in the condition, but this solution will only work for devices connected via station not for devices connected to AP:

`IPAddress SSDPClass::localIP(){
    tcpip_adapter_ip_info_t ip;
    if (WiFi.getMode() == WIFI_STA) {
        if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip)) {
            return IPAddress();
        }
	} else if (WiFi.getMode() == WIFI_AP_STA) {	              // added workaround
		if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip)) {
			return IPAddress();
		}
    } else if (WiFi.getMode() == WIFI_OFF) {
        if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip)) {
            return IPAddress();
        }
    }
    return IPAddress(ip.ip.addr);
}
`

luc-github, 2019-04-25:
WIFI_AP_STA is normaly used for being in AP mode and scan others AP 
Nothing work well in this mixed mode,  I can be wrong as I do not use this mode

Anyway because answer must go to same  direction as query, you need to detect from which IP the query come:
if come from AP connection the answer must be AP IP
if come from STA connection the answer must be the STA IP

Now if you says this code work - I trust you

```
if (WiFi.getMode() == WIFI_AP_STA) {	// added workaround if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip)) { return IPAddress(); } }
```

but may I know why you use this mixed mode?  and for what application ?
Is webserver properly working on both mode AP and STA ?

MiTron66, 2019-04-26:
My Intention is to have access to the the webserver via two ways STA if I am at home having my router available, AP if there is no connection to the router possible. A parallel connection is not intended.
In both modes the webserver is working fine.

luc-github, 2019-04-26:
That is common usage to do and that perfectly make sense - but how do you end up in mixed mode?
 you should be in STA if connection is Ok and AP if no connection, not in AP_STA mode

closed 2020-02-06 by stale[bot]