luc-github/ESP32SSDP #4: [FEATURE REQUEST] Need the ability to add sub elements to <device>

f34rdotcom, 2020-02-03:
Need the ability to add sub elements to \<device\> such as \<serviceList\>

luc-github, 2020-02-04:
an entry point could be added in `<device> I think, so you can add what ever you want to add

f34rdotcom, 2020-02-12:
Since the class holds all of the critical data maybe it would be easier to just have the ability to specify a template and only provide a basic schema template if any at all.

luc-github, 2020-02-12:
yes as schema part will be rewrite to generate string according built functions

luc-github, 2020-02-15:
Still travelling so not full time on this - also I am thinking on how to allow full  template without having multiple string copies to keep low footprint and good performances

luc-github, 2020-02-21:
It should be done soon - I am back home and can test code easily now

luc-github, 2020-02-28:
in  https://github.com/luc-github/ESP32SSDP/commit/0c30711d751e0c5ef1d2d7de4230016450de4358

I think memory footprint could be improved if use String instead of fixed char array, but current version is Ok for the moment

luc-github, 2020-02-28:
I have added also icons list:
`void seticons(const char * icons)` 
Set icons list, NB: optional, this is ignored under windows 
like: 
```  
SSDP.seticons(	"<icon>"
			"<mimetype>image/png</mimetype>"
			"<height>48</height>"
			"<width>48</width>"
			"<depth>24</depth>"
			"<url>icon48.png</url>"
			"</icon>");
  ```
 
for services:
`setservices(const char * services)`
 Set service list, NB: optional for simple device   
like         
```
     SSDP.setservices("<service>"
				"<serviceType>urn:schemas-upnp-org:service:SwitchPower:1</serviceType>"
				"<serviceId>urn:upnp-org:serviceId:SwitchPower:1</serviceId>"
				"<SCPDURL>/SwitchPower1.xml</SCPDURL>"
				"<controlURL>/SwitchPower/Control</controlURL>"
				"<eventSubURL>/SwitchPower/Event</eventSubURL>"
				"</service>");
```

luc-github, 2020-03-11:
@f34rdotcom I guess this one can be closed too ?

closed 2020-04-04 by stale[bot]