luc-github/ESP32SSDP #5: [FEATURE REQUEST] Need ability to set uuid

f34rdotcom, 2020-02-03:
The uuid needs to be able be changed from its default.

luc-github, 2020-02-04:
current UID is based on MAC address but but can also be hardcoded, in that case it will be programmer responsibility to be sure all devices have unique uuid
I can add a function or define for this

f34rdotcom, 2020-02-04:
I am ok with MAC address for suffix to keep it unique. Technically just the
prefix should be user defined.

luc-github, 2020-02-04:
ho ok

luc-github, 2020-02-21:
I am back home now so I will focus on SSDP open issues,I will try to close all before end of next week

luc-github, 2020-02-28:
done in https://github.com/luc-github/ESP32SSDP/commit/0c30711d751e0c5ef1d2d7de4230016450de4358
add `setUUID(const char * uuid, bool rootonly = true);`
default is root only
like `SSDP.setUUID("daa26fa3-d2d4-4072-bc7a");`
it will use  38323636-4558-4dda-9188-cda0e6 + 4 last bytes of mac address if not set
    
Allows also  to set full UUID (not only root)  like 
`SSDP.setUUID("daa26fa3-d2d4-4072-bc7a-a1b88ab4234a", false); `

f34rdotcom, 2020-03-01:
Looks good. Tested.

closed 2020-03-01 by f34rdotcom