luc-github/ESP32SSDP #6: [FEATURE REQUEST] Need ability to change SERVER product name and version.

f34rdotcom, 2020-02-03:
From the spec for the SERVER product token RFC docs.
  "Server implementors are encouraged to make this field a configurable option."

It is partially a security issue but it also allows me to more clearly identify the service.
https://www.w3.org/Protocols/HTTP/1.1/rfc2616bis/draft-lafon-rfc2616bis-03.html#rfc.section.14.38

luc-github, 2020-02-04:
this part is in web server not in SSDP from what I understand , or I misunderstand `protocol HTTP`?

f34rdotcom, 2020-02-04:
Ya, it is all a blur what is what with SSDP :) on its own its just UDP discovery but the headers are rfc HTTP headers. https://github.com/luc-github/ESP32SSDP/blob/4ac65d895986e73389e801a0fe20e6db3f8dc556/ESP32SSDP.cpp#L55

When I implement the HTTP(S) server code it would report similar SERVER header at least for the software and version.

luc-github, 2020-02-04:
ok can be added for packet nb then

luc-github, 2020-02-15:
void setServerName(const char *name); function will be added, name string will cover name and version, default will be "Arduino/1.0" if not set

luc-github, 2020-02-28:
done in https://github.com/luc-github/ESP32SSDP/commit/0c30711d751e0c5ef1d2d7de4230016450de4358

f34rdotcom, 2020-03-01:
Looks good. Tested.

closed 2020-03-01 by f34rdotcom