luc-github/ESP32SSDP #7: [FEATURE REQUEST] need ability to set modelDescription

f34rdotcom, 2020-02-03:
I use modelDescription for the code I am porting. It is not critical but seems to be one of the more common device elements.

luc-github, 2020-02-04:
I was not aware of this section - will test and check

f34rdotcom, 2020-02-04:
Well a new problem. I am using https://github.com/fhessel/esp32_https_server and it does not a typical Client class. This makes using ::schema() to build a schema xml block impossible. 
https://github.com/fhessel/esp32_https_server/blob/f9513ac86e4c5a5b53e6e3258d1061b72cf8efab/examples/Static-Page/Static-Page.ino#L106
I would need a string pointer returned from an override for ::schema()

luc-github, 2020-02-04:
I think I did it for async before
 , I need to check later

f34rdotcom, 2020-02-04:
Thanks for checking. For now I just added a temp function override for schema() to return a char * so I can keep testing :)

luc-github, 2020-02-04:
using async I was building my own string for the webserver handler out of SSDP library actually :
https://github.com/luc-github/ESP32SSDP/blob/master/ESP32SSDP.cpp#L64-L106

I think a rewrite to implemente other featrure request will need to use same mecanism but this time inside of SSDP library and a char * to static builded string will be the way I will go

luc-github, 2020-02-04:
ok found the entry in doc - not sure why it is not in this schema, it is the ported schema from esp8266 SSDP library - will add it and corresponding function this week, still travelling now and not easy to setup env for testing

f34rdotcom, 2020-02-05:
No worries not rushing :) I also noted last night that UPC is common but I
have never seen anything use it.

luc-github, 2020-02-15:
`void setServerName(const char *name);` function will be added, name string will cover  name and version, default will be "Arduino/1.0" if not set

luc-github, 2020-02-28:
in https://github.com/luc-github/ESP32SSDP/commit/0c30711d751e0c5ef1d2d7de4230016450de4358

f34rdotcom, 2020-03-01:
Looks good. tested.

closed 2020-03-01 by f34rdotcom