luc-github/ESP32SSDP #8: [BUG] M-Search reply ST header does not conform to specification.

f34rdotcom, 2020-02-05:
Section 1.2.3 swww.upnp.org/specs/arch/UPnP-arch-DeviceArchitecture-v1.0-20080424.pdf

"To be found, a device must send a UDP response to the source IP address and port that sent the request to the multicast channel. Devices respond if the ST header of the M-SEARCH request is “ssdp:all”, “upnp:rootdevice”, “uuid:” followed by a UUID that exactly matches one advertised by the device."

It will respond with the device type instead.
https://github.com/luc-github/ESP32SSDP/blob/4ac65d895986e73389e801a0fe20e6db3f8dc556/ESP32SSDP.cpp#L210

I suggest just keeping track of the ST and returning it "if" a matching device is found. This also is an issue with the current logic on match device types I will discuss in a feature request to replay to ssdp:all and rootdevice as if they are the same as asking for the actual device type.

f34rdotcom, 2020-02-05:
Just echos back the same header.
https://github.com/nutechsoftware/alarmdecoder-webapp/blob/bb5f0329fc33af534e14ccc37f957a5f22718202/ad2web/discovery.py#L132

luc-github, 2020-02-05:
As this is just a porting of ESP8266 code and it was doing what I needed, I never really had to check specifications - I will have a look

f34rdotcom, 2020-02-08:
Added to PR #10

luc-github, 2020-02-11:
I think this one can be closed - or still something need to be done ?

f34rdotcom, 2020-02-11:
All fixed thanks!

closed 2020-02-11 by f34rdotcom