luc-github/ESP32SSDP #9: [FEATURE REQUEST] Respont to multiple device types or callback?

f34rdotcom, 2020-02-05:
Currently the API accepts a single device type. UPNP clients normally look for a root device then read device xml. The xml contains the real device type but the search target header is 
"ST: upnp:rootdevice"

So setting the device SSDP.setDeviceType() to say upnp:rootdevice works for discovery to start but fails because the xml device info schema will contain the wrong device type.
\<deviceType\>upnp:rootdevice\</deviceType\>

In my case my actual device type is "urn:schemas-upnp-org:device:Basic:1" but I need to respond to "upnp:rootdevice" and "ssdp:all" also the search term reply header is supposed to be the same as the ST in the search regardless of the actual device type. 

Maybe this is best handled in a callback of some type? The ability to respond to multiple device types is a good feature but in most cases just one is enough provided the above rootdevice and ssdp:all are also returning a valid reply.

luc-github, 2020-02-05:
to be honest I did the porting 2 years ago without any knowledge of upnp protocol, I just wanted to get the icon of ESP32 in window just like for ESP8266 ^_^
I will need some time to refresh my brain on this code,but  I think it  should not be a big issue to respond to multiple device types. 🤞

luc-github, 2020-02-21:
what about a dynamic array builded by  `void addDeviceType(const char *deviceType);` where  `void setDeviceType(const char *deviceType);` would become an alias for backward compatibility
each call to the function would add it to the list

and `void removeDeviceType(const char *deviceType);` and `void clearDeviceType();` would handle the list content