luc-github/ESP32SSDP #17: [BUG] compiler warning - large integer implicitly truncated to unsigned type

datacute, 2020-08-16:
Using platformio to compile https://github.com/sharandac/My-TTGO-Watch (which uses this library), gives a warning:

.pio\libdeps\ttgo-t-watch\ESP32SSPD\ESP32SSDP.cpp: In constructor 'SSDPClass::SSDPClass()':
.pio\libdeps\ttgo-t-watch\ESP32SSPD\ESP32SSDP.cpp:112:19: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     _notify_time(0)
                   ^

This is actually because of the earlier line in the constructor:
    _interval(SSDP_INTERVAL),

Where _interval is uint8_t and SSDP_INTERVAL is 1200

Should _interval be uint32_t or unsigned long instead?

luc-github, 2020-08-16:
yes indeed should be uint32_t

luc-github, 2020-08-16:
fixed - thank you https://github.com/luc-github/ESP32SSDP/commit/803e713c0a4a132f2db7ea317d2e83f02a990024
^_^

closed 2020-08-16 by luc-github