luc-github/ESP32SSDP #20: [Question]<Enter comprehensive title>

MitchBradley, 2020-09-08:
Why is the name in library.properties "ESP32SSPD" instead of "ESP32SSDP" (last two characters transposed)?

luc-github, 2020-09-09:
Good catch, it is there since the begining, some dyslexia happen sometimes

Fixed - thank you ^_^

luc-github, 2020-09-10:
I think issue can be closed now

closed 2020-09-10 by luc-github

MitchBradley, 2020-09-10:
The new name does not appear in the platformio library registry.  You might need to do a "platformio library publish" to make the new name show up.

luc-github, 2020-09-10:
Interresting I never did any publishing of this library - you mean it is part of platformIO already ?
how do you set this ?

In Marlin Firmware for esp32 
I have just added the entry :
 `ESP32SSDP=https://github.com/luc-github/ESP32SSDP.git` https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini#L360

MitchBradley, 2020-09-10:
I don't have any experience with this so I am just going by the documentation at
https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing

Your library is already in the registry under the transposed name.  You can find it with the `pio lib search` command using either of the names:
```
PS C:\Users\wmb\Documents\GitHub\Grbl_Esp_New> pio lib search ESP32SSDP
Found 1 libraries:

ESP32SSPD
=========
#ID: 6066
Simple SSDP library for ESP32. Only for ESP32

Keywords: communication
Compatible frameworks: Arduino
Authors: Me-No-Dev, luc-github

PS C:\Users\wmb\Documents\GitHub\Grbl_Esp_New> pio lib search ESP32SSPD
Found 1 libraries:

ESP32SSPD
=========
#ID: 6066
Simple SSDP library for ESP32. Only for ESP32

Keywords: communication
Compatible frameworks: Arduino
Compatible platforms: Espressif 32
Authors: Me-No-Dev, luc-github
```

You can also find it via the platformio web page search:

https://platformio.org/lib/search?query=ESP32SSDP
or
https://platformio.org/lib/search?query=ESP32SSPD

In a platformio.ini lib_deps spec, the only form that currently works is  "ESP32SSPD", as in
```
[env]
lib_deps =
	TMCStepper@>=0.7.0,<1.0.0
        ESP32SSPD
```

luc-github, 2020-09-10:
Weird doc still mention git possibility
https://docs.platformio.org/en/latest/projectconf/index.html

so
```
lib_deps =
TMCStepper@>=0.7.0,<1.0.0
https://github.com/luc-github/ESP32SSDP.git
```
should work

be sure you deleted your `.pioenv` and your `.piolibdeps` I got some trouble in past when modifying platformIO.ini but not cleaning these directories

luc-github, 2020-09-10:
I have just test and 
```
lib_deps =
TMCStepper@>=0.7.0,<1.0.0
https://github.com/luc-github/ESP32SSDP.git
```
works on my side on latest Devt of GRBL_ESP32 - with rlibrary removed from libraries directory of course

luc-github, 2020-09-10:
```
[env]
lib_deps =
	TMCStepper@>=0.7.0,<1.0.0
	ESPSSDP=https://github.com/luc-github/ESP32SSDP.git
```
works also