luc-github/ESP32SSDP #21: [BUG] icon list is outside of device tag

CriPstian, 2020-09-17:
**Describe the bug**
Studying the universal plug and play device arhitecture [HERE](https://openconnectivity.org/upnp-specs/UPnP-arch-DeviceArchitecture-v2.0-20200417.pdf) on page 48-49. I noticed why my details from description.xml is not displayed correctly. The `iconList` attribute is outside of the `device` attribute. That causes various clients to not be able to process the data that we need to share.

**To Reproduce**
The basic example present inside the project is enough to prove my point.
Take a look using [WiFiman](https://play.google.com/store/apps/details?id=com.ubnt.usurvey)

**Expected behavior**
Details regarding my device to be displayed correctly since I comply with the protocol

**Firmware:**
- ESP core version: espressif32@1.12.4   arduino@1.0.4
- Library Version: 1.1.0
- Wifi mode: STA / AP

**Board used (please complete the following information):**
 - MCU: ESP32
 - Name: esp32doit-devkit-v1
 - Flash size: 16M

**Additional context**
The following xml is provided

Following is what I get using the basic example, but fixed it by modifying the library you awesomely provided
```
<root>
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<URLBase>http://192.168.0.118:80/</URLBase>
<device>
<deviceType>my device name</deviceType>
<friendlyName>my device name</friendlyName>
<presentationURL>/</presentationURL>
<serialNumber>123456789</serialNumber>
<modelName>my model name</modelName>
<modelDescription>a model description</modelDescription>
<modelNumber>AA:BB:CC:11:22:33</modelNumber>
<modelURL>http://example.com</modelURL>
<manufacturer>a manufacturer</manufacturer>
<manufacturerURL>https://www.manufacturer.com</manufacturerURL>
<UDN>uuid:322345342-444s44-1242-812juinflsd</UDN>
<serviceList>
<service>
<serviceType>urn:schemas-upnp-org:service:SwitchPower:1</serviceType>
<serviceId>urn:upnp-org:serviceId:SwitchPower:1</serviceId>
<SCPDURL>/SwitchPower1.xml</SCPDURL>
<controlURL>/SwitchPower/Control</controlURL>
<eventSubURL>/SwitchPower/Event</eventSubURL>
</service>
</serviceList>
<iconList>
<icon>
<mimetype>image/png</mimetype>
<height>48</height>
<width>48</width>
<depth>24</depth>
<url>icon48.png</url>
</icon>
</iconList>
</device>
</root>
```
What you render is (take a look at the **iconList** location (is outside of the device tag):
```
<root>
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<URLBase>http://192.168.0.108:80/</URLBase>
<device>
<deviceType>my device type</deviceType>
<friendlyName>a very friendly name</friendlyName>
<presentationURL>/</presentationURL>
<serialNumber>1234567890</serialNumber>
<modelName>a model name</modelName>
<modelDescription/>
<modelNumber>AA:BB:CC:11:22:33</modelNumber>
<modelURL>https://www.model-url.com</modelURL>
<manufacturer>the manufacturer</manufacturer>
<manufacturerURL>https://www.manufacturer.com</manufacturerURL>
<UDN>uuid:322345342-444s44-1242-812juinflsd</UDN>
<serviceList/>
</device>
<iconList/>
</root>
```

luc-github, 2020-09-17:
PR merged -  good catch and thank you

closed 2020-09-17 by luc-github

CriPstian, 2020-09-18:
@luc-github in which release will this be available for **PlatformIO**?

luc-github, 2020-09-18:
I never did any realease for the moment
for platformIO you just need to add the git to` lib_dep` as mentioned in docs: https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

```
lib_deps =
https://github.com/luc-github/ESP32SSDP.git
```

It works for me

CriPstian, 2020-09-18:
Ok thanks!