luc-github/ESP32SSDP #26: [BUG] Prevent std::bad_alloc when udp.parsePacket() is called

wladwnt, 2021-09-14:
When SSDP library is used on ESP32 with other software that consumes lots of RAM, then it might happen that WiFiudp library tries to allocate 1460 Bytes using new operator (in parsePacket() function), but it throws an error (std::bad_alloc), that lead to reboot of ESP32. The reboot can be prevented if in ESP32SSDP.cpp the line number 322:
```c++
nbBytes= _server->parsePacket();
```
replaced by:
```c++
try {nbBytes= _server->parsePacket();} catch (const std::bad_alloc&) { return;}
```

luc-github, 2021-09-14:
a try / catch should be used  only if no other solution, there should be another solution if allocation of new failed

wladwnt, 2021-09-14:
Why not, this is most simple and efficient way and the only way without changing WiFiudp lib. It does not harm anything. SSDP will still continue working (when next udp packet arrived)

luc-github, 2021-09-14:
I  see there is one `new` that is not checked https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiUdp.cpp#L225
Do you think i could be the root cause?

wladwnt, 2021-09-14:
My traces indicates that it happens rather on line 210: if new operator fails it throws std::bad_alloc.

luc-github, 2021-09-14:
interresting, line 210 if new fail , it should return null and so function should return 0 
what esp32 core version  do you use?

wladwnt, 2021-09-14:
I use esp32 arduino 1.0.6 (based in idf 3.3). In my understanding if new operator is used with standart type, then std::bad_alloc is thrown already during new operator execution, so the program aborts before the return value is checked to be zero or not. So lines 2111-213 do not make sense. They would make sense if new would be used like:
```c++
char * buf = new (std::nothrow) char[1460];
```
But might be I am wrong, I am not that deep in this stuff..

luc-github, 2021-09-15:
Hmm 1.0.6 is known to have several issues actually - I would suggest to try to revert to 1.0.4 or to move to 2.0.0 to see if it can be reproduced.
I never got such issue, but I also never used the 1.0.6 as I have sticked to 1.0.4 for a while,  Now moved to 2.0.0

wladwnt, 2021-09-15:
In my experience 1.0.6 (idf 3.3.5 based) is much better than 1.0.4 (idf 3.2 based), but must be used correctly.
I tried 2.0.0, it does not work for me now (I use lots of different protocolls, some from esp-idf directly and have to rewrite some parts to be used with esp-idf 4.4 that is basis for esp32-arduino 2.0.0).

I think 99% of arduino users dont have esp32 programms that periodically use almost all of RAM, therefore no one faced this problem with UDP.cpp / new  operator.

luc-github, 2021-09-15:
any `new` that failed would raise the error, I do not think it is limited to udp

wladwnt, 2021-09-15:
For sure not, but everything else I use either use malloc() (like AsyncUDP) or (std::nothrow) to handle that cases correctly.
Especially with FreeRTOS it is completly wrong not to handle this case and let bad_alloc exception to be thrown, since just simple solution using vTaskDelay and then retry allocation will do the job in most cases.

Thats why using (actually bad written) Udp lib it is ok, in my opinion, to cath this exception on higher level (=much better than not to do that).

luc-github, 2021-09-16:
ok so your suggestion is to use try/catch  for now,, then rewrite the lib to use asyncUdp for reliability and so lib will become async
right?

wladwnt, 2021-09-16:
Danke!

commit 53b3ddfe (2021-09-16): Change travis for github-action

Add try / catch for  parspacket per @wladwnt suggestion for some edge cases
fixes #26
commit b5fdc005 (2022-03-25): Asyncudp (#30)

* Change travis for github-action

Add try / catch for  parspacket per @wladwnt suggestion for some edge cases
fixes #26

* Update build-ci-dev.yml

* separated http header & changed device types (#28)

* separated http header & changed device types

* corrected constructor and schema

* added async example

* Fix CI broken with new example

* Bumb library version

* Version 1.2.0 using AsyncUdp

* Remove debug

Add missing function for backward compatibility

* Remove unused code

* Move sources to src directory

* Update ESP32SSDP.cpp

Add some sanity check

* Add debug featu [... cut]
closed 2021-09-16 by wladwnt