luc-github/ESP32SSDP #36: [Question]License

yunle-wang, 2023-05-30:
In LICENSE file and project indicated this is GPL-3 project, however in source code, there is MIT in place.
I want to double confirm which LIC is for this project? Thank you

luc-github, 2023-05-30:
original code is mit but my version is gpl3 I will update header

f34rdotcom, 2023-05-30:
my 2c. I can't use GPL3. It is too restrictive for any commercial use.

luc-github, 2023-05-30:
according https://softwareengineering.stackexchange.com/questions/105912/can-you-change-code-distributed-under-the-mit-license-and-re-distribute-it-unde
I should keep MIT and add GPL3 - I checked long time ago but forget it
keep MIT header and adding gpl3 is actually correct

luc-github, 2023-05-30:
@f34rdotcom sorry to read that but all code I do / share should be available if used