luc-github/ESP32SSDP #38: [BUG] SSDP Async dont work in APSTA mode only in STA

JulioCRO, 2023-12-26:
**Describe the bug**
Using the example for the "async" method, I realized that I only get a response when using STA mode, in APSTA hybrid mode, responses are not sent to multicast (unlike what happens on the 8266, I'm migrating to esp32).

**To Reproduce**
I used the async web server example contained in ESPAsyncWebSrv.h and added the basics of the ESP32SSDP.h example (also async)
`//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebSrv.h>
#include "ESP32SSDP.h"

AsyncWebServer server(80);

const char* ssid = "XXXXXXXXXXXXX";
const char* password = "XXXXXXXXXXXXX";

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

void setup() {

    Serial.begin(115200);
    WiFi.mode(WIFI_MODE_APSTA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [] (AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request){
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });
        server.on("/description.xml", HTTP_GET, [&](AsyncWebServerRequest *request) {
            request->send(200, "text/xml", SSDP.schema(false));
        });
    server.onNotFound(notFound);

    server.begin();

            //set schema xml url, nees to match http handler
        //"ssdp/schema.xml" if not set
        SSDP.setSchemaURL("description.xml");
        //set port
        //80 if not set
        SSDP.setHTTPPort(80);
        //set device name
        //Null string if not set
        SSDP.setName("Philips hue clone");
        //set Serial Number
        //Null string if not set
        SSDP.setSerialNumber("001788102201");
        //set device url
        //Null string if not set
        SSDP.setURL("index.html");
        //set model name
        //Null string if not set
        SSDP.setModelName("Philips hue bridge 2012");
        //set model description
        //Null string if not set
        SSDP.setModelDescription("This device can be controled by WiFi");
        //set model number
        //Null string if not set
        SSDP.setModelNumber("929000226503");
        //set model url
        //Null string if not set
        SSDP.setModelURL("http://www.meethue.com");
        //set model manufacturer name
        //Null string if not set
        SSDP.setManufacturer("Royal Philips Electronics");
        //set model manufacturer url
        //Null string if not set
        SSDP.setManufacturerURL("http://www.philips.com");
        //set device type
        //"urn:schemas-upnp-org:device:Basic:1" if not set
        SSDP.setDeviceType("rootdevice"); //to appear as root device, other examples: MediaRenderer, MediaServer ...
        //set server name
        //"Arduino/1.0" if not set
        SSDP.setServerName("SSDPServer/1.0");
        //set UUID, you can use https://www.uuidgenerator.net/
        //use 38323636-4558-4dda-9188-cda0e6 + 4 last bytes of mac address if not set
        //use SSDP.setUUID("daa26fa3-d2d4-4072-bc7a-a1b88ab4234a", false); for full UUID
        SSDP.setUUID("daa26fa3-d2d4-4072-bc7a");
        //Set icons list, NB: optional, this is ignored under windows
        SSDP.setIcons(  "<icon>"
                        "<mimetype>image/png</mimetype>"
                        "<height>48</height>"
                        "<width>48</width>"
                        "<depth>24</depth>"
                        "<url>icon48.png</url>"
                        "</icon>");
        //Set service list, NB: optional for simple device
        SSDP.setServices(  "<service>"
                           "<serviceType>urn:schemas-upnp-org:service:SwitchPower:1</serviceType>"
                           "<serviceId>urn:upnp-org:serviceId:SwitchPower:1</serviceId>"
                           "<SCPDURL>/SwitchPower1.xml</SCPDURL>"
                           "<controlURL>/SwitchPower/Control</controlURL>"
                           "<eventSubURL>/SwitchPower/Event</eventSubURL>"
                           "</service>");

        Serial.println("Starting SSDP...");
       if ( SSDP.begin()){
        Serial.println("SSDP started");
       }else{
        Serial.println("SSDP Fail");
       }
 //      WiFi.mode(WIFI_MODE_APSTA);
}

void loop() {
}`

1) Changed WiFi.mode(WIFI_MODE_STA); to WiFi.mode(WIFI_MODE_APSTA);

**Expected behavior**
SSDP response sent to broadcast (Show in windows network) in both STA and APSTA modes.
**Screenshots**

**Firmware:**
- ESP core version: v2.0.14
- Library Version:1.2.0
- Wifi mode:APSTA 

**Board used (please complete the following information):**
 - MCU: ESP32
 - Name: esp32dev
 - Flash size: 4MB 1.2M/1.5M

luc-github, 2023-12-27:
Only STA is supported currently : https://github.com/luc-github/ESP32SSDP/blob/master/src/ESP32SSDP.cpp#L156-L170
why using APSTA ?

JulioCRO, 2023-12-28:
> Only STA is supported currently : https://github.com/luc-github/ESP32SSDP/blob/master/src/ESP32SSDP.cpp#L156-L170 why using APSTA ?

Because sometimes I need both working AP and STA, for example, test if i can connect to new network.
But as you pointed maybe just add "or WiFi.getMode() == WIFI_AP_STA" since the code catch IP from STA.
I will perform some tests and gives a feedback

luc-github, 2023-12-28:
if you need to test a new network you can scan from AP mode by switching to APSTA then when you join the network you need to  switch to STA - the servers need to tight to one IP not 2 so you need to choose AP or STA

luc-github, 2024-01-12:
@JulioCRO any update / feedback ? Or can I close issue ?