luc-github/ESP32SSDP #43: [FEATURE REQUEST] Compatible library code

mathertel, 2024-06-16:
Please consider implementing a library version that is compatible with 3.x and pre-3.x.
using if (ESP_ARDUINO_VERSION_MAJOR < 3)

It is hard for typical Arduino users to distinguish between versions so a version 2.0 that runs with both ESP32 board adoptions is prbablx welcome.

e.g.:

      IPAddress SSDPClass::localIP() {

    #if (ESP_ARDUINO_VERSION_MAJOR < 3)
        tcpip_adapter_ip_info_t ip;
        if (WiFi.getMode() == WIFI_STA) {
          if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_STA, &ip)) {
            return IPAddress();
          }
        } else if (WiFi.getMode() == WIFI_OFF) {
          if (tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip)) {
            return IPAddress();
          }
        }
        return IPAddress(ip.ip.addr);
      }
    
    #else
        esp_netif_ip_info_t ip;
        if (WiFi.getMode() == WIFI_STA) {
          if (esp_netif_get_ip_info(get_esp_interface_netif(ESP_IF_WIFI_STA), &ip)) {
            return IPAddress();
          }
        } else if (WiFi.getMode() == WIFI_OFF) {
          if (esp_netif_get_ip_info(get_esp_interface_netif(ESP_IF_ETH), &ip)) {
            return IPAddress();
          }
        }
    #endif
        return IPAddress(ip.ip.addr);
      }

Thanks for your library.

luc-github, 2024-06-17:
thank you is that working with platformio also ? Is this `ESP_ARDUINO_VERSION_MAJOR` present ?

luc-github, 2024-06-18:
Merged thank you

closed 2024-06-18 by luc-github