luc-github/ESP32SSDP #46: [BUG] v2.x branch: invalid IP address used

folkertvanheusden, 2025-05-12:
**Describe the bug**
resource available
  USN:      uuid:38323636-4558-4dda-9188-cda0e6d38000b::upnp:rootdevice::upnp:rootdevice
  Location: http://165.165.165.165:80/description.xml
resource available
  USN:      uuid:38323636-4558-4dda-9188-cda0e6d38000b::upnp:rootdevice::upnp:rootdevice
  Location: http://176.4.0.0:80/description.xml

<root xmlns="urn:schemas-upnp-org:device-1-0">
<specVersion>
<major>1</major>
<minor>0</minor>
</specVersion>
<URLBase>http://36.60.13.128:80/</URLBase>
<device>
<deviceType>urn:schemas-upnp-org:device:rootdevice:1</deviceType>
<friendlyName>Mains frequency</friendlyName>
<presentationURL>/</presentationURL>
<serialNumber>esp32-c4ead380</serialNumber>
<modelName>Mains frequency measurer</modelName>
<modelDescription/>
<modelNumber/>
<modelURL>https://www.vanheusden.com</modelURL>
<manufacturer>vanheusden.com</manufacturer>
<manufacturerURL>https://www.vanheusden.com</manufacturerURL>
<UDN>uuid:38323636-4558-4dda-9188-cda0e6d38000b868</UDN>
<serviceList/>
<iconList>
<icon>
<mimetype>image/png</mimetype>
<height>48</height>
<width>48</width>
<depth>24</depth>
<url>favicon.ico</url>
</icon>
</iconList>
</device>
</root>

These are not valid IP addresses in my lan, they should've been in the 192.168.64.0/23 range.

**Firmware:**
- ESP core version: framework-arduinoespressif32 @ 3.0.0
- Library Version:[e.g 2.1b4]
- Wifi mode: STA  

**Board used (please complete the following information):**
 - MCU: [e.g. ESP32] 
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2025-05-12:
do you have same issue using sample code ?

folkertvanheusden, 2025-05-14:
Hi,

Found the reason: SSDPClass::localIP does not handle the WIFI_AP nor 
WIFI_AP_STA cases. In those cases, esp_netif_ip_info_t (or the v2 
variant tcpip_adapter_ip_info_t) is not initialized and returns garbage.

luc-github, 2025-05-14:
you wrote : 
Wifi mode: STA

Yes I AP mode it make no sense to have SSDP because in AP mode the IP is fixed, unless you use it as gateway but I never tested this configuration.

I will check tcpip_adapter_ip_info_t initialization

folkertvanheusden, 2025-05-14:
Agreed, but in the AP_STA mode it does as there the esp is a dhcp client as well.

luc-github, 2025-05-14:
AP_STA is used only for scan usually in AP Mode or  for gateway -  why do you need AP_STA ?
also if I remember all services like webserver or  others will take first  mode available - so AP - did this changed ?

so currently I only support STA mode

folkertvanheusden, 2025-05-14:
It's just one of the possibilities. I use it with a wifi manager to 
choose at run time between configure or run-mode of the wifi-settings.

luc-github, 2025-05-14:
if in AP mode and switching to AP_STA for scan, the application should go back to AP once scan is done, that is good  practice to save state and put back if any change.

closed 2025-05-14 by luc-github